OpenMYR/IoT_Motors #1: Network Instability when sending UDP command packets

brandonbrown2, 2016-04-02:
"missed packets and high jitter in ping response "
Christopher

"what is the latency to your router vs the motor
i assume the motor is on your network"
Kyle
"ESP12 average 160ms with 2% loss
router is 40ms with 0% loss
statistically better but feels just as bad from the stepper application"
Christopher

Vzor-, 2016-04-03:
This occurs when there is an open tcp connection. our connections need to close sooner, though high latency  with tcp open is still a bug and should be investigated further. Please do not close this issue until both are addressed.

Vzor-, 2016-04-05:
This can wait till 0.3.0 because when #7 is done, tcp connections should only be open for miliseconds. This means the connection will only be unstable for fractions of a second.

closed 2016-06-19 by brandonbrown2