OpenMYR/IoT_Motors #14: wdt reset when changing station mode config while in station mode

brandonbrown2, 2016-04-02:


Vzor-, 2016-04-06:
This issue has become very intermittent but is still an issue.

Vzor-, 2016-04-06:
The device still saves the new network info and after the wdt reset the device behaves as expected. The only symptom seen by the user is if commands are queued up when you change network, it may forget them.

brandonbrown2, 2016-04-06:
Tried disconnecting from the station before switching to the new one, and the crash still occurred.

Next thing to try is closing all TCP and UDP connections, including server, before switching. This could dovetail nicely into using SoftAP+Station as a stopgap between modes, see #9

Vzor-, 2016-06-18:
This issue has not been seen in months and may be gone, testing is needed.

closed 2016-06-19 by brandonbrown2