OpenMYR/IoT_Motors #2: Brushed DC Motor Driver switches direction incorrectly

brandonbrown2, 2016-04-02:
the code seems to have a bug into when it needs to change directions, sometimes it does not drive the pins correctly
you need to make sure the current pin is brought low before changing directions.

but it did seem to do something weird at direction changes like  pin was held high sometimes
the directions are not always being refocused on direction change.

Vzor-, 2016-04-02:
@brandonbrown2 @chradams Showing Brandon how to message more than one person at a time

Vzor-, 2016-06-18:
This product is being put on the back burner, removing from milestone. Due to the radical changes in how the motor drivers are controlled, this issue is out of date and will be closed.

closed 2016-06-18 by Vzor-