OpenMYR/IoT_Motors #19: Some devices can't communicate with the device in SoftAP mode

brandonbrown2, 2016-04-06:
When the device is in SoftAP mode, successful communication with the device seems to be intermittent. It works fine for @Vzor- but I'm getting a self-assigned IP address instead of one from the DHCP server on the ESP8266. If I set a static IP on my computer (ie 192.168.4.2), communication is successful.

If you have a similar issue, please leave your computer's specs here.
Mine are:
Windows 10 Home x64
Intel(R) Dual Band Wireless-AC 3160 PCI\VEN_8086&DEV_08B4&SUBSYS_82708086&REV_93

Vzor-, 2016-04-08:
found others with this issue https://github.com/esp8266/Arduino/issues/1094

Vzor-, 2016-04-09:
fixed as of 396f0db0e83a5b0441cf7f8c1d2f73508b6268e6 in branch 0.1.0-tcp-json

commit 396f0db0 (2016-04-09): Disconnect from wifi station before switch to softAP mode
fixes #19
closed 2016-04-09 by Vzor-